pimoroni/badger2040 #4: Badger2040w RTC example

North101, 2023-03-10:
Might be useful to the document the badger2040w RTC and link to the example
https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pcf85063a/pico_rtc.py

Would it be worth having the micropython `machine.RTC` call the `pcf85063a` functions?
https://docs.micropython.org/en/latest/library/machine.RTC.html

Gadgetoid, 2023-03-10:
I think it’s possible but quite tricky to redirect machine.RTC, since we’d have to patch the relevant files in the port (MicroPython source) itself. I found the dual RTC stuff a bit weird on Inky Frame so maybe it’s worth looking into how tricky.

I should be able to port the stuff I’ve recently written for Inky Frame to Badger. That should cover off most basic RTC use cases, since it gives you a function to wake up after a number of minutes.

Gadgetoid, 2023-03-10:
Okay, some basic RTC functions and docs added here - https://github.com/pimoroni/badger2040/pull/2

If you get the build from GitHub Actions you should be able to give these a try.

bjohas, 2023-03-25:
Hi @North101 and @Gadgetoid ,

How would this example https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pcf85063a/pico_rtc.py work with halting the Badger 2040 W?

I'd like to take a temperature reading every 5 minutes, and shut the Pico off for those periods.

As an alternative, I've tried `machine.deepsleep()`, but the rtc seems to get reset (on battery power). Is that expected?

bjohas, 2023-03-26:
In fact, 
```
display = badger2040.Badger2040()
display.halt()
```
also forgets the rtc setting. If I start the `clock.py` application, close it, and restart, it looks for the time again.

bjohas, 2023-04-01:
https://github.com/pimoroni/pimoroni-pico/issues/449

> The catch is that Badger 2040 doesn't rely on the not-great sleep states of the Pico, but rather cuts the power entirely. Connected battery or not, it'll lose the time since it's truly off. The buttons actually trigger the power being turned back on, in addition to being readable by the RP2040. Rather ingenious save this one unfortunate side-effect.

This explains the behaviour.

thirdr, 2025-01-08:
Closed as resolved.

closed 2025-01-08 by thirdr