pimoroni/badger2040 #7: Badger 2040 W: Refresh occasionally fails on battery

helgibbons, 2023-03-16:
Noticed that the occasional (1 in 10, perhaps) refresh doesn't complete properly when on battery - almost like the board's cutting the screen power early?

Wonky refreshes at 0.03 seconds and 0.19 seconds:
https://user-images.githubusercontent.com/50950368/225606357-c048e0bc-3273-4fb3-81c7-6a8fced7d213.mp4

I'm using this [MicroPython with examples build](https://github.com/pimoroni/badger2040/actions/runs/4425426816) (but it happens with the older pimoroni-pico builds too).

Gadgetoid, 2023-03-20:
Is this on Badger or Badger 2040 W? IIRC I had to put in a timeout since the busy-wait flag was unreliable, I wonder if that's working at all.

I'm seeing some *weird* stuff on the 2040 W with the screen getting stuck on refresh and not responding at all and also what looks to me like the "invert" flag getting set on the screen hardware somehow.

I think I should probably charge my battery...

helgibbons, 2023-03-20:
2040 W!

commit 92e62e0c (2023-03-21): Badger OS: Add keepalive to fix #7.

Holding down a button, or pressing a button with the "right" timing would keep the Badger on, despite the 3v3 en pin being turned off.

This would result in abrupt power loss when the button is released, causing screen corruption.