pimoroni/badger2040 #15: Missing version in the info  launcher?

typxxi, 2023-04-23:
If I start the Badger W I can not find a version from the launcher menu . 
I had expected to find the 0.0.2 mentioned here for the most recent one in the INFO section at least, maybe in the  help section too.

Right now I am asking myself if I had gotten the Badger W shipped last week with V 0.0.2 or not and if I have to update or not.

typxxi, 2023-04-23:
I suggest to simply avoid any confusion in the future by adding the Version number where it natively belongs: 

top line of the launcher which starts with:"BADGER OS" followed by a "bar showing x% used" or so and then on the right corner is enough space for a V0.0.2 or if not there then attached on the left to BADGER OS - V0.0.2 which looks quite good and says everything I can  not figure out now. 

Why ?

I had gotten the badger surprisingly fast for UK royal mail within 8 days in Germany and then instantly started to play around with the screen to get used to its features and quirks , I mean the epaper update weirdness for one who had never owned such a display before. Then I added some missing lines of code to the examples and was quite happy till I asked myself how to update the pico and if it still would need the copy file thing which would erase all my implemented ideas . So I checked this site and found no answer if I have to update or not considering that the V0.0.2 is just 3 weeks old not knowing about the pirates speed to upgrade their production. So far great implementation but I think I need a better WiFi integration cause moving around is a mess cause I can not configure the wifi config file all the time and would need a different approach to switch through all WIFI_CONFIG*.py files once the current WiFi could not be detected or does not work anymore. Would also be a great OS improvement to fix this issue. Right now I had to swap my router from office, project and what not to cell phone cause the cell phone is my companion that always comes along with me.

typxxi, 2023-04-23:
Here is an example that shows the current version in the launcher app upper right corner.

I would also suggest to mention the release date there at least week and year which should make the sporadically user to look for any updates if it might feel too old. Now 0.0.2 can be everything even though i do not know if I am using 0.0.2 on a device shipped from the UK just 2 weeks after release date. At the upper right corner there is enough space for this "V0.0.2 - 10/2023" and it is simple by this line after the current line 114   display.text("badgerOS", 4, 4, WIDTH, 1.0)

display.text("V0.0.X", 266, 4, WIDTH, 1.0)

and for the version with a release week it has to start a little earlier

display.text("V0.0.X - 10/2023", 224, 4, WIDTH, 1.0)

[image]

[image]

typxxi, 2023-04-26:
If you start thonny and check the shell after a boot you should see a kind of release hint cause you get this line

MicroPython 38e7b84-dirty on 2023-03-30

where the date can let you assume that you have gotten the latest version V0.0.2 shipped to me 14th of april or so which would mean that Pimoroni has shipped the badger with the latest OS they had released so far. 

Just found it out from a screenshot - after i had downloaded the official version and suddenly saw the hints and checked my older screenshots. But again: you need the release / version on the top level of the device and not somewhere burried deep and maybe network info might be the right spot or another small app about the badger OS and the memory and storage usage . 

[image]

salimhb, 2023-04-27:
> I need a better WiFi integration

It's another issue, but I hear you. This [repo](https://github.com/tayfunulu/WiFiManager) has a very nice solution, which I think BadgerOS should use. I managed to adapt the same code and run it on my badger. I don't know if the maintainers would be interested in a PR.

GMMan, 2023-06-21:
I disagree that the version should be shown on the launcher because that isn't something most users would need to concern themselves with, at least on a day to day basis. It would make more sense to show it in the info app. There should be an API added to expose version, commit hash, build date etc. in the OS side of things.

Gadgetoid, 2025-01-08:
Added to info in #89 

Version has always been available via `import version` which includes a `BUILD` var containing the full release filename.