pimoroni/badger2040 #26: Reading battery level with Badger 2040W

DuaneKaufman, 2023-06-14:
Dear All,

I have noticed that get_battery_level() doesn't perform a measurement:
`def get_battery_level():
    return 0
    # Battery `measurement`

Is this because the method provides erroneous results? Has anyone found a good way to report battery levels when running on battery?

Sincerely,
Duane

salimhb, 2023-06-17:
It looks like left over Code from the old repository (which was for other devices).
I tried uncommenting the return and couldn't get any result. I'm also interested if anyone has hints in the right direction.

GMMan, 2023-06-21:
For further reading: https://github.com/orgs/micropython/discussions/10421

The current code is for Badger 2040 which uses a standalone RP2040 and has the pins set up a bit differently from how they would be on a Pi Pico.