pimoroni/badger2040 #27: How are images to be loaded onto the Badger2040w?

KnownBySomeone, 2023-06-15:
I have not yet figured out how to load image files onto the Badger2040w and get them to display.

Steps I have taken:
-I read the readme within /examples/image.py """"Images must be 296x128 pixel JPEGs   Create a new "images" directory via Thonny, and upload your .jpg files there."""

-I created what I believe to be a 1bit, 296x128 pixel .jpg image file.  The same file as attached.

-I loaded said file into the /images folder alongside the /images/badgerpunk.jpg file, and got the message, "JPEG: could not read file/buffer. JPEG: could not read file/buffer."

-I have also loaded the image in other locations that did not follow the readme, but have had no success as well.

-I have also separately tried changing which extra data is saved and tried saving the image as grayscale when extracting the image from GIMP.

[image]

helgibbons, 2023-09-04:
Closing for now, but feel free to reopen if you're still having issues.

closed 2023-09-04 by helgibbons

wesbos, 2024-10-30:
I had this issue as well and it turned out it was a progessive jpg