pimoroni/badger2040 #30: Badger 2040 display when connecting

exussum12, 2023-06-29:
https://github.com/pimoroni/badger2040/blob/879f04435ede2b7607522a0c34656debad52451b/firmware/PIMORONI_BADGER2040W/lib/badger2040.py#L242

The 2040w is hard coded to take over the display when connecting, Could this be an option to light the LED or similar user defined action.

Example use case 
Showing latest information from a website could look like

```
while True:
    display.keepalive()
    variables = get_data()
    draw_page(variables)
    sleep_for(10)
```

After waking up from sleep, the wifi needs to be connected again, taking ~10 seconds, so every 10 minutes, for 10 seconds the badge doesnt display the intended content. A user defined option could be doing an update partial to the screen or lighting the LED.

exussum12, 2023-06-29:
https://github.com/pimoroni/badger2040/blob/main/badger_os/examples/news.py#L179

The example news app could (probably) take advantage of sleep if this was implemented

salimhb, 2023-06-29:
I have a similar use case. I'm using [wifimgr.py](https://github.com/tayfunulu/WiFiManager/blob/master/wifimgr.py) and modified it to work on badger. My approach Is to render the text in the top of the screen and then call `display.partial_update(0, 0, WIDTH, 24)`.

I also use the LED as an indicator of activity:
```python
while True:
    display.keepalive()
    display.led(128) # brighter LED
    render() # this does the data fetching and drawing
    display.led(6) # dimmed LED (on USB)
    badger2040.sleep_for(10) # on battery the LED goes completely off during sleep
```

Gadgetoid, 2023-06-30:
That status handler code is quite bad on two accounts- the `print()` is not always desirable either.

Normally this is solved by passing in a custom handler when constructing `NetworkManager`, but this choice has been lost via the badger lib. I'll raise a PR to fix it.

exussum12, 2023-06-30:
Perfect, I have done something similar to @salimhb for now (though still using the config file rather than adding wifi creds in again. 

Would be happy to test anything!

Gadgetoid, 2023-06-30:
Okay, test build here - https://github.com/pimoroni/badger2040/actions/runs/5421334513

Now you can use:

```
display.connect(status_handler=None)
```

Or define your own status handler in user code:

```
def status_handler(mode, status, ip):
        print(mode, status, ip)

display.connect(status_handler=status_handler)
```

exussum12, 2023-06-30:
Yeah works well for me, I have added a comment on the PR though as the method still causes unexpected behaviour.

Thanks for the super quick fix!

exussum12, 2023-07-22:
This was resolved by the PR

closed 2023-07-22 by exussum12