pimoroni/badger2040 #32: Wifi not initializing on Badger2040W

christopherdoyle, 2023-07-08:
I am getting the following error trying to initialize wifi network using `.connect`:

```
ampy.pyboard.PyboardError: ('exception', b'', b'Traceback (most recent call last):\r\n  File "<stdin>", line 19, in <module>\r\n  File "badger2040.py", line 264, in connect\r\n  File "uasyncio/core.py", line 1, in run_until_complete\r\n  File "uasyncio/core.py", line 1, in run_until_complete\r\n  File "uasyncio/core.py", line 1, in run_until_complete\r\n  File "network_manager.py", line 78, in client\r\nOSError: [Errno 1] EPERM\r\n')
```

I have tried this on 3 different wifi networks and with v0.0.1 and v0.0.2 of the firmware, and the firmware [from the latest build](https://github.com/pimoroni/badger2040/actions/runs/5421639672).

The simplest example of the error is:

```
>>> import network
>>> wlan = network.WLAN(network.STA_IF)
>>> wlan.active(True)
>>> wlan.scan()
Traceback (most recent call last):   
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM
```

I thought hardware issue, but then I flashed the latest [stock rp2 pico-w firmware](https://micropython.org/download/rp2-pico-w/), and the same command works correctly and lists wifi networks. I am able to connect with `wlan.connect` and get an ip address.

christopherdoyle, 2023-07-08:
Specifically clocking to "fast" seems to fix this, doesn't really explain the difference between badger2040w firmware and pico-w firmware though.

```python
badger2040.system_speed(badger2040.SYSTEM_FAST)
```

Gadgetoid, 2023-07-28:
Relates to WiFi simply breaking at slower clock speeds which is mentioned here - https://github.com/pimoroni/badger2040/pull/25

closed 2025-01-08 by thirdr