pimoroni/badger2040 #50: Display banding when power is removed

G8PTN, 2023-09-14:
I am using a Badger 2040 with what I believe is the latest firmware.
Filename: pimoroni-badger2040-v0.0.3-micropython-with-badger-os.uf2

I have noticed that there is sometimes a vertical banding (lines) on the display after power is removed.

This appears to have been raised before, but the implication is that it was fixed with a firmware update.
https://github.com/pimoroni/pimoroni-pico/issues/704
https://forums.pimoroni.com/t/display-banding-artefacts-on-badger-2040w-display/21521

Are there any know causes for this effect?
Regards, Dave

exussum12, 2023-09-18:
This also appears to be temperature related, when the display is cold, this happens much less

goozgig, 2023-09-20:
Returned 1 unit due to 'banding' (occurred after less than 1 week occasional operation during test based on examples etc).

Replacement now demonstrating similar, again after about a week.

1. Tried latest and previous firmware
2. Banding appears approx 1 minute after power removal (USB or VSys), or deepsleep.
3. Could it be associated with power circuit??

Comments/suggestions/solution from Pimoroni Support (or others) would be appreciated, thank you.

exussum12, 2023-09-20:
Has anyone had this issue when not displaying an image? I can only make mine do it when showing an image on the screen

G8PTN, 2023-09-20:
Hi All,
I have not seen the banding effect after setting the screen to all white, but have not really done extensive testing.
Unfortunately the effect is not very repeatable. I would say most of the time it is okay.

Generally it seems to occur after the unit has been powered for a long period and then depowered. This may be related to the observation that it appears to be temperature related.

If the effect does not occur within a couple of minutes of depowering, it then seems to be okay, so this would be okay for a static display. I did wonder if there was some strange rectification occurring, which was somehow sufficient to charge the E-ink display.

In some of the other forums, it implies that it does not occur if the battery is left connected. I am not sure why this would be the case, since the power to the RP2040 and E-ink display automatically turns off with the default Badger OS and the current drawn from the battery is approximately 3uA (LDO regulator).

My understanding is that the external circuit associated with the E-ink display (MOSFET, Inductor and diodes) is used to generate a nominal +/- 20V. I don’t have any experience with E-ink displays, but I would expect as long as the firmware driver follows the required operational flow for the display, it should be in an off state.

On one of the other forums, it implies that there was a firmware fix for the Badger 2040W.

From an interest point of view it would be nice to understand the cause and what the firmware fix was for the 2040W version.

Regards, Dave

goozgig, 2023-09-24:
... thanks everyone.

I don't recall experiencing banding from a blank screen, but I will add that although mostly vertical, I have also seen (very thin/feint) horizontal lines.
During recent days it has been much less frequent. The battery (3 x AA, VSys and deepsleep) is now 3.15V having started this run at 3.91V. Interestingly, operating temperature has declined from about 23 to 20-21C.

Conclusion so far: effect marginal, probably influenced by screen content, power supply, temperature, and component tolerances.

Following on from G8PTN, it is tempting to consider power down sequencing/timing, e.g. between 3V3 and VGL/VGH (influenced by C2/C4 and switched by GDR)??
I can't see how GDR  is under Pico program control/firmware updates, unless someone can correct me, please?

G8PTN, 2023-10-13:
Another minor observation.

After running the clock example, as well as the occasional 
[image]
banding effect, I also noticed that where the clock least significant second digit had been, there was a shadow of this when showing a blank all-white screen (see attached).

After repeatedly re-writing the all-white screen, the shadow of the clock image did eventually fade away.

This may just be a side effect of how the partial update for the clock screen works and is not related to the banding issue.

Arthur-Philip-Dent, 2023-11-12:
The clock is a really good example to sentence a e-ink display to death... (the shadow is wear-off, imho not the same like banding). 

I think I tracked banding down to vagabonding radio signals effecting some column lines as well some row lines (funnily depending on the orientation of the badger). 

I replaced a e-ink display on my badger to see the very same effect with "banding" (maybe we should call this effect better spider silk), which disappears completly after a refresh, to come back after a (undeterminded) while. 

After I shielded the flex pcb, which connects the display with the Badger pcb, the spooky spider silk was gone...

[image]
[image]

exussum12, 2023-11-12:
Nice work @Arthur-Philip-Dent!

goozgig, 2023-11-12:
... interesting: Arthur-Philip-Dent ... was the foil screen connected to 0V (or other?).

Also (latest firmware), since original observations in Aug and unusually warm Sep, now with cooler weather the banding has not manifested itself !!

(More preoccupied at the moment trying to figure out cause of 'badger2040.sleep_for(sleep_mins)' failing to sleep, often after more than 12 hours of reliable operation powered by battery --- but that is a different issue !!).

Thanks ...

Arthur-Philip-Dent, 2023-11-13:
Grounding the shield to 0V: not connected. Aluminum is barely not solderable. I had no copper foil. Wrapped it around the connector, where on the sides it is surface mounted, but there’s for sure not connection. Shielding for magnetic fields works anyway.

goozgig, 2023-11-13:
.... Arthur-Philip-Dent: I wonder if tight-wrapping foil around ribbon cable results in some capacitative coupling between conductors and the shield, leading to modification of signal waveforms/timings, sufficient to 'filter' the problem away? Another marginal effect?
(Somewhat unscientific speculation without proper measurements with fast 'scope, tuned probes and rf sniffer etc ...!!).

Doesn't explain how 'lines' can appear 1 min or so after power removal, which makes me suspect display power cct.

As stated, my display is now behaving, but I'll have a go with screening if/when artifacts re-manifest ... thanks, G.

G8PTN, 2023-11-14:
Interesting observations. I tend to agree with the comments from goozgig that it could imply marginal timings and it does not really explain why the lines appear after the power is removed.

goozgig, 2024-02-05:
... further investigations with battery-powered Badger (ref: Sep 24 2023).

Since banding tended to occur after approx 30 secs to 1 min or so, and suspecting something to do with power-down sequencing, I monitored PicoW pin 36 (3V3 OUT) with a digital multimeter as power was removed from 'batt' connector.
It took approx 20 secs+ for the nominal 3.3 volts to decline to near zero (No longer having access to oscilloscopes etc. prevents more accurate measurements!).

By way of speculation, I attached a 4K7 resistor between 3V3 and ground, conveniently making use of pins on the BME280 QWIIC-connected sensors board (VIN and GND see photo).

The 3V3 collapsed within 1 or 2 seconds of power removal.
Banding has not been seen since (4 weeks), - battery-powered system that wakes up every 5 minutes, samples pressure, humidity and temperature, updates display then sleeps,  thereby powering up/down 12 times/hour.
-A few tests with USB power also failed to show banding-

Hope others find this useful ... g

[image]

G8PTN, 2024-02-08:
Hi,
Thanks very much for the update.

For completeness I have measured the 3V3 rail on the Badger2040. Due to the integrated discharge function in the AP22802 load switch which feeds the power to the RP2040, the decay time is fast (see attached).

[image]

The power supply arrangement on the Badger2040W is completely different. There is no discharge function and the 3V3 is derived from a switched mode power supply on the PICO W, which has some high value smoothing capacitors. This would explain the long decay time on the 3V3 rail you measured and may be the cause of the banding issue.

Unfortunately, It still does not really explain the banding problem on the Badger2040.
Regards, Dave

goozgig, 2024-02-12:
**Badger 2040W**

... thanks Dave - good to see the the 3V3 fall as a measurable 'scope trace,  - and ... (at this late stage) -  making me aware of key differences between Badger 2040 and its 'W' cousin !!

I have been puzzling and commenting on the banding issue for months, without realising/considering the very significant differences between Badger 2040 and 2040W - so obvious now ! Having only a '2040W, I was assuming -wrongly- that differences were confined to Pico/PicoW boards, and haven't been making it clear that my banding was a 'W' version.

However, the ePaper power circuits do appear to be identical for both variants. If probe(s) could be attached, it would be interesting to see traces for VGH and VGL, and possibly elsewhere, to look for any coincidences with the appearance of banding.

(Happily, still no banding with the 3V3 leak resistor on my '2040W - for whatever reason ! - marginal timing effect?).

73s Gerry.