pimoroni/badger2040 #51: white space is mashed for bitmap fonts

yjeanrenaud, 2023-09-14:
I would like to reopen this issue. The whitespace character is messing things up somehow. See:
[image]
I used ` badger.text(strLine, x, y, scale=2,spacing=1,fixed_width=True)` (yeah I know spacing is not a thing when using the bitmap fonts, but I tried to make sure)
If I use _ for padding up spaces (using micropythons .format) everything is as expected, but space characters (ASCII 0x20 I suppose) fail to do so. Will have a look at the code if I got some time left

yjeanrenaud, 2023-09-14:
https://github.com/pimoroni/pimoroni-pico/pull/846/commits/91f95a2e006d7c25e9df6cbeb4d1b66a16159188
PR is done for pimoroni-pico, so badger2040 needs a version upped?

Gadgetoid, 2023-09-18:
Fix should be available in CI builds for this PR - https://github.com/pimoroni/badger2040/pull/52

helgibbons, 2024-06-21:
Think this one is now sorted, closing for now.

closed 2024-06-21 by helgibbons