pimoroni/badger2040 #57: Missing CmakeLists.txt

waterbury, 2023-10-17:
I'd like to build the uf2 from source, but there is a lack of a CMakeLists.txt file, and build instructions.

paulinder, 2023-10-17:
hi - i'm not a dev on this project, but might i suggest you try following the steps in the [github actions workflow](https://github.com/pimoroni/badger2040/blob/main/.github/workflows/micropython.yml) file? that's where the build happens.

marc-git, 2024-11-22:
I would like to second this request. Although you are perhaps targetting users who will mostly just run your uf2 files I think in the spirit of the open source project here and of Pimoroni itself, it would be a minimum to provide build instructions. As @paulinder suggests, I need to basically hack your github workflow to compile the uf2 file with a modified micropython