pimoroni/badger2040 #60: Badger_OS v0.0.4 does not recognize 'WL_GPIO2' on Badger2040W

Arthur-Philip-Dent, 2023-11-03:
Same code from #59 on same Badger2040W with Badger_OS [v0.0.4 ](https://github.com/pimoroni/badger2040/releases/tag/v0.0.4) :

```
from machine import ADC, I2C, Pin

# reading 'WL_GPIO2' on a picoW tells us whether or not USB power is connected (VBUS Sense)
vbus = Pin("WL_GPIO2", Pin.IN).value()
print(f"vbus: {vbus}")

```

I personally find this one the more challenging output: 
```
>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
ValueError: unknown named pin "WL_GPIO2"
>>> 
```

Did I mess something up or are the releases getting some "winter tuning" lately? 

Need a towel?

The Hitchhiker

helgibbons, 2023-11-06:
Hello!

Seems like 'EXT_GPIO2' works for the VBUS sense pin with v0.0.4, but 'WL_GPIO2' does not - not sure why that might be 🤔 Paging @Gadgetoid ?

Gadgetoid, 2024-02-27:
Day late and a dollar short here, but I think this is a bug in MicroPython's CMakeLists here:

https://github.com/micropython/micropython/blob/8fdcc25eb07731a64077b7576c5d477cf81e2dd5/ports/rp2/CMakeLists.txt#L529

We're supplying an explicit `MICROPY_BOARD_DIR` to build Badger 2040W with an out-of-tree board definition which lives here - https://github.com/pimoroni/badger2040/tree/main/firmware/PIMORONI_BADGER2040W

But the above code assumes `MICROPY_BOARD_DIR` == `MICROPY_BOARDS_DIR/MICROPY_BOARD` which is not always the case.

I'll raise a patch for this, but it'll probably be a while before that fix makes its way to Badger.

Gadgetoid, 2024-02-27:
Okay after digging my way through multiple levels of code broken by bleeding-edge MicroPython I managed to produce a build and test that my fix actually works- thankfully it does:

```
MicroPython a14d36d-dirty on 2024-02-27; Pimoroni Badger2040W 2MB with RP2040
Type "help()" for more information.
>>> from machine import Pin
>>> vbus = Pin("WL_GPIO2", Pin.IN).value()
```

If you *really need* "WL_GPIO2" (granted quite some time has passed since you raised this) then you can grab a test build here: ⚠️  https://github.com/pimoroni/badger2040/pull/74 ⚠️ 

But *BEWARE* this test build will trash your Badger filesystem. Even if you grab the one without Badger OS. It may also have multiple other bugs caused by bleeding edge MicroPython that we haven't yet found/fixed.

Gadgetoid, 2024-06-19:
Newer version is imminent (I've been working on it this e... morning.. night?) and it will add `WL_GPIO2` in addition to a bunch of other useful pin definitions.

The fix for this ended up being applied *after* MicroPython v1.23.0 was released 😭 so I ended up having to make a patch anyway.

WIP is now in #75