pimoroni/badger2040 #61: OSError: [Errno 110] ETIMEDOUT

optio, 2023-11-04:
Hi,

Since recently, I receive the folllowing error, using badger2040w , pimoroni-badger2040w-v0.0.4-micropython-with-badger-os.uf2

Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
  File "badger2040.py", line 58, in <module>
OSError: [Errno 110] ETIMEDOUT

The error is thrown at this line:

`i2c.writeto_mem(0x51, 0x00, b'\x00')  # ensure rtc is running (this should be default?)`

https://github.com/pimoroni/badger2040/blob/a8552a55ad090783f9bf06aa2a07821051accf87/firmware/PIMORONI_BADGER2040W/lib/badger2040.py#L57C31-L58C88

Any idea how to solve this? I tried reflashing pimoroni-badger2040w-v0.0.4-micropython-with-badger-os.uf2 though even with the default launcher, the error is thrown.

helgibbons, 2023-11-06:
Sounds like the RTC might has got into a weird state - you could try unplugging the battery and leaving it for little while to allow it to reset?

optio, 2023-11-07:
I left badger2040w unplugged for 2 days , though still same error. Will try again in a week or so...

helgibbons, 2023-11-08:
That should be sufficient for the RTC to reset.

You could try running this RTC example to see if it unsticks things? https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pcf85063a/pico_rtc.py

You could also try clearing Badger's flash memory, as described here: https://learn.pimoroni.com/article/getting-started-with-badger-2040#troubleshooting

optio, 2023-11-12:
I first put flash_nuke.uf2 , then after restart put pimoroni-badger2040w-v0.0.4-micropython.uf2.

When running https://github.com/pimoroni/pimoroni-pico/blob/main/micropython/examples/pcf85063a/pico_rtc.py I get:

`>>> %Run -c $EDITOR_CONTENT`
`MPY: soft reboot`
` `
`PCF85063A rtc demo!`
`===================`
`initialise rtc.. `

which does not continue further, no error shown either.

When I added a print after PimoroniI2C()

`PINS = {"sda": 4, "scl": 5}  # i2c pins 4, 5 `
`i2c = PimoroniI2C(**PINS) `
`print("i2c ok") `

the print 'i2c ok' does not show, thus it's already stuck on `i2c = PimoroniI2C(**PINS) `

Btw Inky Frame 4 connects just fine, so it's not a thonny/cable/... issue.

helgibbons, 2023-11-13:
Does flashing our standard Pico W build and then re-running the RTC example help?

https://github.com/pimoroni/pimoroni-pico/releases

optio, 2023-11-15:
Thanks for the help, though still no luck.

I first reflashed 'flash_nuke.uf2' , then flashed 'pimoroni-picow-v1.21.0-micropython.uf2' , though the pico_rtc.py example still failed at `rtc.datetime(now)` .

Initially I though the pico_rtc.py sample failed earlier, based in which print() statement did last output, though upon removing/adding line by line in , it was clear that `PimoroniI2C()` seems to work fine, also `rtc = PCF85063A(i2c)` , though hangs when calling `rtc.`

I tried inputting the script manually using the Thonny shell, where the shell stops responding upon `rtc.datetime(now)` or ` rtc.reset()` or any call to `rtc.`.
After a few minutes, I click the stop button , which results in `Unable to connect to ...`. A reset makes the badger2040 responsive again.

Could it be the wrong PINS values?

```python
>>> import machine
from pimoroni_i2c import PimoroniI2C
from pcf85063a import PCF85063A, MONDAY
import time
led = machine.Pin(22, machine.Pin.OUT)
led.high()
PINS = {"sda": 4, "scl": 5}  # i2c pins 4, 5
i2c = PimoroniI2C(**PINS)
rtc = PCF85063A(i2c)
>>> rtc
<pcf85063a>
>>> rtc.reset()
─────────────────────
Unable to connect to COM3: could not open port 'COM3': PermissionError(13, 'Access is denied.', None, 5)
If you have serial connection to the device from another program, then disconnect it there first.
Process ended with exit code 1.
```

helgibbons, 2023-11-17:
Thanks for that - the RTC example runs as is on my Badger W so the pins should be correct.

I suspect you may have a faulty RTC if trying to initialise it hard crashes the Pico W - [drop a note to support](https://pimoroni.freshdesk.com/support/tickets/new) with a link to this issue and they should be able to sort you out with a replacement Badger.

optio, 2023-11-21:
Thanks for the debugging support @helgibbons .
A new badger2040w is shipped to me within warranty by Pimoroni.

closed 2023-11-21 by optio