pimoroni/badger2040 #67: Badger 2040W: Date_Time?

PieterJanSterk, 2023-12-22:
Hi, 

I wrote a little script that looks at a dictionary with birthday and shows me how long until the next person has a birthday. 

```
import badger2040
import utime

# Birthday information
birthdays = {
    'Person1': (9, 1),
    'Person2': (20, 1),
    'Person3': (4, 2),
}

# Function to get the current date in the format DD/MM
def get_current_date():
    current_time = utime.localtime()
    current_day = current_time[2]
    current_month = current_time[1]
    return current_day, current_month

# Function to display birthday message
def display_birthday_message(display, next_birthday_person, days_until_birthday):
    # Display "Happy birthday!" on the screen
    display.set_pen(15)  # White background
    display.clear()

    display.set_pen(0)  # Black text
    display.set_font("bitmap8")

    # Display the name centered, larger, and a bit lower
    text_width = display.measure_text(next_birthday_person, scale=5)
    display.text(next_birthday_person, (296 - text_width) // 2, 30, scale=5)

    # Display "Happy birthday!" centered and larger
    birthday_message = "Gelukkige verjaardag!"
    text_width = display.measure_text(birthday_message, scale=3)
    display.text(birthday_message, (296 - text_width) // 2, 95, scale=3)

    display.update()
    utime.sleep(5)  # Display the message for 5 seconds

# Function to calculate countdown
def countdown(current_date, stop_date):
    current_day, current_month = current_date
    stop_day, stop_month = stop_date

    # Calculate days until the next birthday
    if current_month < stop_month or (current_month == stop_month and current_day <= stop_day):
        days_until = (stop_month - current_month) * 30 + (stop_day - current_day)
    else:
        # Birthday is in the next year
        days_until = (12 - current_month + stop_month) * 30 + (stop_day - current_day)

    return days_until

# Display Setup
display = badger2040.Badger2040()

# Run the script every 8 hours
while True:
    # Get the current date
    current_date = get_current_date()

    # Find the next birthday
    next_birthday_person = None
    days_until_birthday = float('inf')

    for person, birthday in birthdays.items():
        days_until = countdown(current_date, birthday)

        if days_until < days_until_birthday:
            days_until_birthday = days_until
            next_birthday_person = person

    # Set pen color and font for the next person's birthday and days until
    display.set_pen(15)  # White background
    display.clear()

    display.set_pen(0)  # Black text
    display.set_font("bitmap8")

    # Display the next person's birthday centered, larger, and a bit lower
    text_width = display.measure_text(next_birthday_person, scale=5)
    display.text(next_birthday_person, (296 - text_width) // 2, 30, scale=5)

    # Display remaining days larger and centered
    days_text = f"Nog {days_until_birthday} dagen"
    text_width = display.measure_text(days_text, scale=4)
    display.text(days_text, (296 - text_width) // 2, 95, scale=4)  # Updated height to 95

    display.update()

    # Check if it's someone's birthday
    if days_until_birthday == 0:
        display_birthday_message(display, next_birthday_person, days_until_birthday)
    
    # Calculate and display countdown
    count_days = days_until_birthday
    countdown_text = f"Countdown: {count_days} days"
    text_width = display.measure_text(countdown_text, scale=4)
    display.text(countdown_text, (296 - text_width) // 2, 130, scale=4)
    
    display.update()

    # Sleep for 8 hours
    utime.sleep(1 * 60)  # 8 hours in seconds
```

Runs fine with Thonny! But when I run this on battery, the date is complely wrong on screen. Does this script use date time from Thonny? How do you get the correct date from the onboard 2040 chip, untethered?

salimhb, 2023-12-22:
I think that you should first connect to wifi and get the date from NTP. Check the example [clock.py](https://github.com/pimoroni/badger2040/blob/a8552a55ad090783f9bf06aa2a07821051accf87/badger_os/examples/clock.py#L12-L20)

https://github.com/pimoroni/badger2040/blob/a8552a55ad090783f9bf06aa2a07821051accf87/badger_os/examples/clock.py#L12-L20

PieterJanSterk, 2024-01-05:
Thank you. Sorry for asking, this is all a bit new to me. Thanks!

closed 2024-01-08 by PieterJanSterk