pimoroni/badger2040 #72: Badger 2040W Won't Connect to Unsecured Wireless

pottsrichard, 2024-02-13:
Our school uses heavily filtered, unsecured wireless - mainly for any non-staff devices. 

Our recently bought Badger 2040Ws don't want to connect to the unsecured WiFi (standard Raspberry Pis connect fine). We've tried several variations in the WIFI_CONFIG.py file, but without any luck.

We've tried:
PSK = ""
PSK = None
Adding a line with: KEY_MGMT = None
Replacing the PSK line with: KEY_MGMT=None

I've tried taking a look at net_info.py, but it only seems to pass the SSID and PSK from WIFI_CONFIG.py

Any ideas guys?

goozgig, 2024-02-22:
Hi,

Put together this simple wifi connect function whilst implementing an MQTT channel on my Badger 2040W (with very useful info from: https://core-electronics.com.au/guides/getting-started-with-mqtt-on-raspberry-pi-pico-w-connect-to-the-internet-of-things/).

Never tried connecting to open networks, but this would allow multiple connect tries and reporting, which should help with finding the solution.

Good luck .... Gerry

Code________________________________________

import network
from time import sleep

ssid = 'myssid'
password = 'mypassword' # or whatever required for open network ???

def connect_to_wifi(max_connection_tries):
    print("*** connecting to network ***")
    wlan = network.WLAN(network.STA_IF)
    wlan.active(True)
    wlan.connect(ssid, password)    
    for i in range(0, max_connection_tries):
        connect_tries = i
        print("** network connection try: " + str(connect_tries))
        if not wlan.isconnected():   
            connect_status = 'Not Connected'
            connection = (0,0,0,0)
            print('... Waiting for connection ... try#: ' + str(connect_tries))
            sleep(1)
        else:              
            connection = wlan.ifconfig()
            connect_status = 'Connected'
            break
    return connection, connect_status, connect_tries

# Test
connection_attempts = 20 # or whatever ...
connection, connect_status, connect_tries = connect_to_wifi(connection_attempts)
print(connection)
print(connect_status)
print(connect_tries)

Gadgetoid, 2024-02-22:
There was a bug in MicroPython's handling of open wireless networks, discussed and eventually fixed here - https://github.com/micropython/micropython/issues/9016

Badger 2040W isn't building against the latest version of MicroPython yet, so it's very possible this could be the problem.

pottsrichard, 2024-02-23:
> There was a bug in MicroPython's handling of open wireless networks, discussed and eventually fixed here - [micropython/micropython#9016](https://github.com/micropython/micropython/issues/9016)
> 
> Badger 2040W isn't building against the latest version of MicroPython yet, so it's very possible this could be the problem.

Thanks Gadgetoid - I'd read that thead when researching the issue and assumed the latest version of Badger 2040W would have this - good to know that a fix should eventually come. Thanks!

BenjaminEHowe, 2024-07-21:
Looks like [v0.0.5](https://github.com/pimoroni/badger2040/releases/tag/v0.0.5) might fix this -- @pottsrichard, could you verify?