pimoroni/badger2040 #76: Screen won't refresh and error code [ERRNO1] EPERM

thedeto84, 2024-03-05:
I think I may have broken the Badger 2040. Getting the error message [ERRNO1] EPERM. 

I have reinstalled the OS and Thonny recognises the device giving the message below.

**MPY: soft reboot
MicroPython v1.21.0-dirty on 2023-10-11; Pimoroni Badger2040 2MB with RP2040
Type "help()" for more information.
>>>** 

Any ideas?

Gadgetoid, 2024-03-06:
At an absolute wild, wild guess I suspect filesystem corruption. Did you reinstall from the -with-badger-os version of the .uf2?

thedeto84, 2024-03-08:
Hi - Yes I reinstalled the badger OS version.uf2 a few times and did the nuke uf2 also. 
I noticed that the black block below is chipped and the exposed copper wire is showing. This is probably the issue maybe? Do you know what this part does?

[image]