pimoroni/badger2040 #77: Badger W

Crispymug, 2024-04-02:
My Badger W won't appear in Thonny. I've reflashed the firmware multiple times and even performed a flash nuke then a reflash but the Badger won't appear as a Serial device and thonny isn't able to connect to the Badger.
Is it an issue with the device or with how I'm handling the software?

Gadgetoid, 2024-04-05:
Might be worth also trying “flash_nuke.uf2” (https://cdn-learn.adafruit.com/assets/assets/000/099/419/original/flash_nuke.uf2?1613329170 ) which will completely erase the flash and eliminate any startup scripts blocking Thonny. In theory the “with_badger_os” builds should do this.

I ran into a similar problem testing some new builds and improvements. It seems catching Badger at the right time can be tricky - I find hitting reset on the board and hitting reset in Thonny quickly after helps.

Might be a good idea for us to implement an escape key- a button you hold down to prevent the Badger software from launching at startup and potentially blocking Thonny.

Crispymug, 2024-04-06:
I've tried the flash nuke and have just tried the rest on both the Badger and thonny.
[image]
Thonny simply can't detect the badger as a device and windows can't either. I've tried the included cable and a few known good data cables and it still doesn't work.
What steps do I take now?

Crispymug, 2024-05-06:
I'm assuming my badge is just a brick then. Thanks for attempting to help.

helgibbons, 2024-05-07:
Sorry for the lack of reply here - [drop us a support ticket](https://pimoroni.freshdesk.com/support/tickets/new) with a link to this issue if you haven't already and we'll get a return/replacement sorted for you.

chrissyhroberts, 2024-07-09:
Just adding to this thread, I had a similar issue on a specific Mac. I couldn’t get Thonny to connect to either of two Badgers running OS above 0.3. Both would connect to Thonny when running 0.3.

Moving to another `clean` mac and also to a PC I was able to connect to badger with 0.5

Going back to my problem Mac, I found that by resetting Thonny helped. 
I deleted the library folder with  `rm -rf ~/Library/Thonny` and this fixed the issue

I was then able to connect badger with OS 0.5

diegomarino, 2024-08-12:
Hi there,

I've been suffering the same issue and to solve it I juse followed two of the suggested steps:
1) Delete the library folder with `rm -rf ~/Library/Thonny`
2) Install the latest release marked `with-badger-os`
3) Restart the computer...

My env is a Macbook Air M2 running `OSX 14.0` and `Thonny 4.1.4`

If I experience again the error, I'll share with you the `backend.log`

diegomarino, 2024-08-13:
Just happened again... Sharing my `backend.log`. The board becomes unresponsive when it's connected to Thonny (trying both Settings>>>Interpreter `Raspberry Pi Pico` and `RP2020` detecting the port automatically)

```
12:44:43.818 [MainThread] INFO    thonny: Thonny version: 4.1.4
12:44:43.818 [MainThread] INFO    thonny: cwd: /Users/diego
12:44:43.818 [MainThread] INFO    thonny: original argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python', '-s', '-u', '-B', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': False, 'port': '/dev/cu.usbmodem1101', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
12:44:43.818 [MainThread] INFO    thonny: sys.executable: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/bin/python3.10
12:44:43.818 [MainThread] INFO    thonny: sys.argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040/rp2040_back.py', "{'clean': False, 'port': '/dev/cu.usbmodem1101', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
12:44:43.818 [MainThread] INFO    thonny: sys.path: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/rp2040', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python310.zip', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/lib-dynload', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages']
12:44:43.818 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=1, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
12:44:43.818 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': False, 'port': '/dev/cu.usbmodem1101', 'dtr': None, 'rts': None, 'submit_mode': 'raw_paste', 'interrupt_on_connect': True, 'write_block_size': 64, 'write_block_delay': None, 'proxy_class': 'RaspberryPiPicoBackendProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
12:44:43.819 [MainThread] DEBUG   thonny.plugins.micropython.serial_connection: Opening serial port /dev/cu.usbmodem1101
12:44:48.461 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 64, write_block_delay: 0, read_block_size: 0
12:44:48.462 [MainThread] INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type RP2040Backend
12:44:48.464 [MainThread] INFO    thonny: TIME/MODS 4.666 before prepare (+73 modules)
12:44:48.464 [MainThread] INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.misc_utils', 'thonny.plugins', 'thonny.plugins.backend_config_page', 'thonny.plugins.micropython', 'thonny.plugins.micropython.bare_metal_backend', 'thonny.plugins.micropython.connection', 'thonny.plugins.micropython.mp_back', 'thonny.plugins.micropython.mp_common', 'thonny.plugins.micropython.mp_front', 'thonny.plugins.micropython.serial_connection', 'thonny.plugins.micropython.webrepl_connection', 'thonny.roughparse', 'thonny.running', 'thonny.tktextext', 'thonny.ui_utils', 'thonny.workdlg', 'tkinter', 'tkinter.commondialog', 'tkinter.constants', 'tkinter.dialog', 'tkinter.filedialog', 'tkinter.font', 'tkinter.messagebox', 'tkinter.simpledialog', 'tkinter.ttk', 'urllib', 'urllib.parse', 'zlib']
12:44:48.464 [MainThread] DEBUG   thonny.plugins.micropython.bare_metal_backend: _process_until_initial_prompt, clean=False
12:44:48.465 [MainThread] INFO    thonny.plugins.micropython.mp_back: Sending interrupt
12:44:48.465 [MainThread] DEBUG   thonny.plugins.micropython.mp_back: Sending ctrl chars: b'\x03'
12:44:48.465 [MainThread] ERROR   thonny.plugins.micropython.mp_back: Exception in MicroPython main method
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/serial/serialposix.py", line 621, in write
    n = os.write(self.fd, d)
OSError: [Errno 6] Device not configured

[... cut ...]
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/micropython/serial_connection.py", line 88, in write
    size = self._serial.write(data)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/serial/serialposix.py", line 655, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 6] Device not configured

```