pimoroni/badger2040 #82: Clock Example does not run on Badger 2040 W

ulrischa, 2024-06-06:
The clock Example does not run on the Wireless Version. Even if I change all references to badger2040w I get errors like AttributeError: 'module' object has no attribute 'is_wireless'. The docs here seems wrong: https://github.com/pimoroni/badger2040/tree/main/badger_os#clock

helgibbons, 2024-06-07:
The [clock example](https://github.com/pimoroni/badger2040/blob/main/badger_os/examples/clock.py) seems to run fine for me on Badger W - do you have the most recent version of Badger W flavoured MicroPython installed?

https://github.com/pimoroni/badger2040/releases/latest

(and install instructions: https://github.com/pimoroni/badger2040 )

ulrischa, 2024-06-07:
Yes habe these but I need to change all badger2040 to badger2040w in the code to run the examples. Also some time related libraries are not found

helgibbons, 2025-03-11:
Hopefully you're now sorted, if not please let us know!

closed 2025-03-11 by helgibbons