pimoroni/badger2040 #85: Can not back out of menu item after selected, deleting "state" file folders is the only way back to the menu

easchaefer, 2024-06-23:
When I choose a selection from the menu, like "badge" it opens the badge, but then, no matter what I do I can't get back to the menu. The only solution I have found is to open the file system via thonny and delete the contents of the "state" folder. Then I reboot and can back to the main menu... until I choose something new, then I am stuck on that option. 

I have flashed it and loaded the latest update, but nothing seems to work. 

I mainly use it as a badge so it's not the biggest deal since I tend to leave it there anyway, but it is super annoying.

Has anyone else seen this issue?

rgsteele, 2024-06-23:
@easchaefer Nothing happens when you press the **A** and **C** buttons at the same time?

easchaefer, 2024-06-23:
@rgsteele 

Well, I feel stupid :) that works.

I always used the "reset" button to go back. I didn't know you could do that. THANKS!

closed 2024-06-23 by easchaefer