pimoroni/badger2040 #93: GitHub Universe 2040 badge

North101, 2024-12-20:
The GitHub Universe 2024 badge is apparently being shipped without firmware and no link to download the firmware. [experimental/rp2350](https://github.com/pimoroni/badger2040/actions/runs/10921237699) branch works however the artifacts have expired. If this branch was tagged and released it would at least give the users a way to load the firmware

unnamedd, 2024-12-21:
@North101 have you checked this GitHub organization? 
https://github.com/badger

They do have everything related to the Badger there and also, they are using MicroPython, therefore, you just had to download and flash a MicroPython firmware for RP2350 (which is the one used in the GitHub Badger), the rest, I am quite sure will be work nicely. Also, if you connect your Badge to you computer and use vìper-ide.org, you should see the code of your badge in there.

North101, 2024-12-21:
I was aware of that repo but I don't have the badge myself. I've just seen at least 2 people ask about it and I'm unable to find a precompiled version of the firmware. I had a quick look again and I don't see one. If you can point me to one I'll know where to point people in the future and I could make a PR to make it more visible

helgibbons, 2024-12-23:
The Github Universe folks have a precompiled .uf2 for Badger 2350 [here](https://github.com/badger/badgerbodger/blob/main/uf2-images/pimoroni-badger2350-6ea88e080bc48350eb48c6608f67c18497709a31-micropython.uf2). I don't think it includes Badger OS though, so you'd also need to copy across [the examples from this repo](https://github.com/pimoroni/badger2040/tree/main/badger_os).

closed 2025-01-08 by thirdr